m5stack/M5Unified #131: Use Speaker and Mic at the same time

yujonglee, 2024-11-11:
https://github.com/m5stack/M5Unified/blob/cf1aec372b840093811f18b10bf3a8a13ffd33e1/examples/Basic/Microphone/Microphone.ino#L38-L40

Not super familiar with this space, but I am curious what's blocking us to use it together.
ESP32S3 has 2 I2S controller, and there's [some comments](https://www.reddit.com/r/esp32/comments/19dfdsr/i2s_mic_and_speaker_simultaneously/) saying it worked.

Any comments on this would be super helpful.

lovyan03, 2024-11-11:
Hello, @yujonglee 
This is because there were no cases where speakers and microphones were used simultaneously when the development of M5Unified started.

We are considering this as a future issue, but if you want to use speakers and microphones simultaneously right now, you will need to implement it yourself without using M5Unified.

yujonglee, 2024-12-06:
Hi @lovyan03, thank you for the comment!

> We are considering this as a future issue

Do you have a rough timeline in mind for when this might happen?

yujonglee, 2024-12-06:
@lovyan03  Thank you for the comment!

Based on the schematic, it seems like single I2S channel is shared.
Could you kindly confirm that the implementation would be using single I2S as full-duplex mode?

[image]
[image]
[image]

lovyan03, 2024-12-08:
Thank you for the information.

I understand everything, but the current implementation does not support full-duplex mode, so if you need the functionality immediately, please use the ESP-IDF I2S driver yourself.

ainyan03, 2026-08-17:
On most boards the speaker and the microphone share a single I2S bus, so M5Unified operates them exclusively: starting one stops the other. This is a hardware constraint rather than a library limitation. If your application needs both, alternate between them (e.g. record, then play back).