m5stack/M5Unified #132: Error with 0.2.0

mgrouch, 2024-11-12:
````
m5atomS3-lite/bbn-m5atomS3-lite/bbn_m5atomS3_lite_DS18B20/build/libraries/M5Unified/utility/Mic_Class.cpp.o
2024-11-12T17:35:33.1045084Z /home/runner/Arduino/libraries/M5Unified/src/utility/IMU_Class.cpp: In member function 'bool m5::IMU_Class::begin(m5::I2C_Class*, m5::board_t)':
2024-11-12T17:35:33.1046789Z /home/runner/Arduino/libraries/M5Unified/src/utility/IMU_Class.cpp:94:76: error: 'board_M5AtomS3RCam' is not a member of 'm5::board_t' {aka 'lgfx::boards::board_t'}
2024-11-12T17:35:33.1047979Z          if (board == m5::board_t::board_M5AtomS3R || board == m5::board_t::board_M5AtomS3RCam || board == m5::board_t::board_M5AtomS3RExt)
2024-11-12T17:35:33.1048668Z                                                                             ^~~~~~~~~~~~~~~~~~
2024-11-12T17:35:33.1049870Z /home/runner/Arduino/libraries/M5Unified/src/utility/IMU_Class.cpp:94:120: error: 'board_M5AtomS3RExt' is not a member of 'm5::board_t' {aka 'lgfx::boards::board_t'}
2024-11-12T17:35:33.1051038Z          if (board == m5::board_t::board_M5AtomS3R || board == m5::board_t::board_M5AtomS3RCam || board == m5::board_t::board_M5AtomS3RExt)
````

lovyan03, 2024-11-12:
Hello, @mgrouch 
probably you need update M5GFX v0.2.0

mgrouch, 2024-11-12:
Yes. With M5GFX v0.2.0 it works. This can be closed.

closed 2024-11-12 by mgrouch