m5stack/M5Unified #135: error: Invalid pin specified in fasted_esp32.h and clockless_rmt_esp32.h

konacurrents, 2024-12-03:
# Using 2.0.1 of M5Unified
I cannot get the FastLED to compile using the M5AtomS3.

```
In file included from /Users/scott/Documents/Arduino/libraries/FastLED/src/platforms/esp/32/fastled_esp32.h:12,
                 from /Users/scott/Documents/Arduino/libraries/FastLED/src/platforms.h:36,
                 from /Users/scott/Documents/Arduino/libraries/FastLED/src/FastLED.h:54,
                 from /Users/scott/Documents/Arduino/libraries/M5AtomS3/src/./utility/LedDisplay.h:4,
                 from /Users/scott/Documents/Arduino/libraries/M5AtomS3/src/M5AtomS3.h:6,
                 from /Users/scott/Documents/GitHub/M5Unified-ESP_IOT_Nov24/ESP32_M5Unified/ESP_IOT/Defines.h:467,
                 from /Users/scott/Documents/GitHub/M5Unified-ESP_IOT_Nov24/ESP32_M5Unified/ESP_IOT/src/SensorClass/KeyUnitSensorClass.h:3,
                 from /Users/scott/Documents/GitHub/M5Unified-ESP_IOT_Nov24/ESP32_M5Unified/ESP_IOT/src/SensorClass/KeyUnitSensorClass.cpp:2:
/Users/scott/Documents/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.h: In instantiation of 'class ClocklessController<23, 72, 72, 144, (EOrder)66, 0, false, 5>':
/Users/scott/Documents/Arduino/libraries/FastLED/src/chipsets.h:605:7:   required from 'class SK6812Controller<23, (EOrder)66>'
/Users/scott/Documents/Arduino/libraries/FastLED/src/FastLED.h:109:52:   required from 'class SK6812<23, (EOrder)66>'
/Users/scott/Documents/Arduino/libraries/FastLED/src/FastLED.h:304:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = SK6812; unsigned char DATA_PIN = 23; EOrder RGB_ORDER = (EOrder)66]'
/Users/scott/Documents/GitHub/M5Unified-ESP_IOT_Nov24/ESP32_M5Unified/ESP_IOT/src/SensorClass/KeyUnitSensorClass.cpp:124:54:   required from here
/Users/scott/Documents/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.h:358:46: error: static assertion failed: Invalid pin specified
     static_assert(FastPin<DATA_PIN>::validpin(), "Invalid pin specified");
                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
```

lovyan03, 2024-12-08:
Hello, @konacurrents 
Sorry, but I don't think the problem is related to M5Unified...

konacurrents, 2024-12-08:
The `M5AtomS3` source directly calls the `LedDisplay` as the error above shows.

So is this just an error with `M5AtomS3` versus `M5Unified`?

Any insight would be helpful.
thanks.

lovyan03, 2024-12-17:
The part causing the error does not appear to be under the jurisdiction of the M5Unified repository.
Try building a combination of M5Unified and FastLED and see if an error occurs. (Not M5AtomS3)

closed 2025-03-16 by lovyan03