m5stack/M5Unified #146: TOF example

smebberson, 2025-01-31:
The [TOF unit information page](https://docs.m5stack.com/en/unit/tof) references the following example from the [M5Stack library](https://github.com/m5stack/M5Stack/blob/master/examples/Unit/ToF_VL53L0X/ToF_VL53L0X.ino). However, this library is now outdated and as you know, doesn't work with ESP32-S3 based M5Stack controllers.

Is it possible to include a working TOF example in this repository that would work with both ESP32, ESP32-Pico and ESP32-S3/ESP32-C* boards?

lovyan03, 2025-01-31:
Hello, @smebberson 
We need to be very careful about including sample code for specific optional products in this repository.
If we start doing that, we will have to include all products here, so we think it is best to basically do this for libraries for each unit.

We are currently working on a separate project called `M5UnitUnified` to unify the unit libraries.
This will likely be where we provide and support the new libraries.

smebberson, 2025-01-31:
@lovyan03, thanks for getting back to me. That makes a lot of sense :)

I've just discovered https://github.com/m5stack/M5Unit-TOF, which seems to use M5UnitUnified.

It looks as though [this example code using UnitUnified](https://github.com/m5stack/M5Unit-TOF/blob/main/examples/UnitUnified/UnitToF/PlotToSerial/main/PlotToSerial.cpp), could be adapted to recreate https://github.com/m5stack/M5Stack/blob/master/examples/Unit/ToF_VL53L0X/ToF_VL53L0X.ino and should work with ESP32-S3 based devices due to using M5Unified?

lovyan03, 2025-01-31:
Yes, that's it.

It should probably work with ESP32S3 and C3 as well.

GOB52, 2025-01-31:
@smebberson 

Hi,  I am the author of the M5UnitUnified.

Cores and Units We are in progress for most combinations.
Also, the unit side libraries are newly written, not external ones.

Already CoreS3, AtomS3, NanoC6... + ToF/ToF4M combinations are working.

If you have any questions, problems, or corrections, we would be happy to receive an Issue/PR in the repository concerned.

smebberson, 2025-02-02:
Hi @lovyan03 and @GOB52, I've made an example that you might like to consider at https://github.com/m5stack/M5Unit-TOF/pull/1.

I was thinking it could be linked on the [TOF documentation page](https://docs.m5stack.com/en/unit/tof) so that S3 based devices users wouldn't attempt to load the [existing example](https://github.com/m5stack/M5Stack/tree/master/examples/Unit/ToF_VL53L0X) and be disappointed to find it doesn't work on their device.

closed 2025-05-16 by lovyan03