m5stack/M5Unified #148: Need a way to force board detection

pguyot, 2025-02-01:
If PSRAM and ESP32 v3 (>1 as a workaround for PSRAM issues) are activated in `idf.py menuconfig`, M5StackCore2 is mistaken for another product (M5TimerCam).

We should be able to override board using menuconfig.

lovyan03, 2025-02-04:
Sorry, I don't really understand what you're talking about, so please explain it again assuming I know nothing.

pguyot, 2025-02-04:
Actually, I was wrong, I thought the issue came from using idf.py menuconfig, but the problem is there with default configuration.

Using master (0.2.3, 7e840c14913d877b8786e7228c66a0100b42bcfd), on a M5StackCore2, after having called `M5.begin()`, `M5.getBoard()` returns `m5::board_t::board_M5TimerCam`.

Is it a bug with the detection?

pguyot, 2025-02-04:
The autodetection mechanism wasn't able to initialize spi bus.
I believe this may be related to release/5.4 being used.
The following patch fixes detection. It probably needs some `#if` or to be rewritten by zero-ing the configuration structure.

```diff
diff --git a/src/lgfx/v1/platforms/esp32/common.cpp b/src/lgfx/v1/platforms/esp32/common.cpp
index c713080..7d183ba 100644
--- a/src/lgfx/v1/platforms/esp32/common.cpp
+++ b/src/lgfx/v1/platforms/esp32/common.cpp
@@ -517,6 +517,7 @@ namespace lgfx
         buscfg.max_transfer_sz = 1;
         buscfg.flags = SPICOMMON_BUSFLAG_MASTER;
         buscfg.intr_flags = 0;
+        buscfg.data_io_default_level = 0;
 #if defined (ESP_IDF_VERSION_VAL)
   #if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 2, 0))
         buscfg.isr_cpu_id = ESP_INTR_CPU_AFFINITY_AUTO;
```

lovyan03, 2025-02-05:
Thank you for the explanation.
I've just updated the develop branch of M5GFX, so please give it a try if you'd like.

pguyot, 2025-02-05:
Thank you. I confirm it works with :
- M5GFX develop (e4dffd0c1659bf1a4b825ade48a74f60dcbfb4ac)
- M5Unified master (7e840c14913d877b8786e7228c66a0100b42bcfd)
- esp-idf release/v5.4 (473771bc14b7f76f9f7721e71b7ee16a37713f26)

I let you close this issue at your convenience.

lovyan03, 2025-03-16:
The fix is ​​in the current release, so I will close this issue.
Thanks for your cooperation !

closed 2025-03-16 by lovyan03