m5stack/M5Unified #156: Wrong values using Rtc.GetTime() [M5Dial]

SadE54, 2025-03-15:
Hi,

I'm using Rtc from M5unified to replace millis() for better precision countdown (my application is a pomodoro timer)

[image]

[image]

But I have inconsistant data . Some time the GetTime(&rtc_time_t) returns false, so I can filter
But sometimes it returns true AND the data are wrong.

[image]

[image]

[image]

PS : Using Platformio with Arduino / M5Unified 0.2.5

lovyan03, 2025-03-16:
Hello, @SadE54 

Is your program multi-threaded and is it possible that another task is using I2C communication?
If so, communication may conflict and data may be corrupted.

Also, we do not recommend frequently obtaining RTC time information during runtime. It is better to access the RTC immediately after startup or once every few hours to synchronize the ESP32 built-in clock and obtain time information from the built-in clock normally.

SadE54, 2025-03-16:
Hi, 

No I'm only use the M5dial without any devices, and in a single thread. And according to the schematic , RTC is the only device on I2C. What is strange is that the wrong values when GetTime returns true, are really similar each time 🤔

ainyan03, 2026-08-27:
Sorry for the long wait — this is addressed in M5Unified 0.2.20 (current release is 0.2.21), so please update.

`getDateTime()` used to decode the RTC's BCD registers without any validation, so a corrupted I2C read produced an impossible time while still returning `true` — matching your "returns true but the values are wrong" case (the similar-looking garbage each time is typical of a read that fails the same way). Since #313 the decoded fields are range-checked (seconds/minutes/hours, day 1–31, month 1–12, weekday) and the BCD digits validated; an implausible read now leaves your `rtc_time_t` untouched and returns `false`, so your existing filter on the return value is enough.

This makes the bad values detectable rather than explaining why the read glitched in the first place (on the M5Dial the RTC is the only device on that bus, so an occasional I2C error is the most likely source). If you still see wrong values with `true` on 0.2.20 or later, please reopen.