m5stack/M5Unified #158: Arduino 3.2rc2 + M5Unified: crash at boot: E (1003) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver

mhaberler, 2025-03-24:
M5Unified/develop at a2753f8, [Pioarduino Arduino 3.20 rc2](https://github.com/pioarduino/platform-espressif32/releases)

````c++
#include <M5Unified.h>
void setup() {
    M5.begin();
}
void loop() {
}
````
Platformio.ini:
```ìni
[env:m5stack-coreS3]
monitor_filters = esp32_exception_decoder
monitor_speed = 115200
platform = https://github.com/pioarduino/platform-espressif32/releases/download/54.03.20-rc2/platform-espressif32.zip

framework = arduino
board = m5stack-cores3
lib_deps =
	https://github.com/m5stack/M5Unified.git#develop

build_type = debug
build_flags =
	-g -O2
	-DCORE_DEBUG_LEVEL=4
````

compile, flash - console log:

````
mode:DIO, clock div:1
load:0x3fce2820,len:0x118c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc20
load:0x403cb700,len:0x30e0
entry 0x403c88b8
E (1003) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver

abort() was called at PC 0x4202ce7f on core 0

Backtrace: 0x40377209:0x3fceb1e0 0x4037c839:0x3fceb200 0x40383431:0x3fceb220 0x4202ce7f:0x3fceb2a0 0x4200e666:0x3fceb2c0 0x403769b6:0x3fceb2f0 0x403ccbd5:0x3fceb340 0x403ccf91:0x3fceb380 0x403c8925:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
  #0  0x40377209 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:454
  #1  0x4037c839 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:92
  #2  0x40383431 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:38
  #3  0x4202ce7f in check_i2c_driver_conflict at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2c/i2c.c:1723
  #4  0x4200e666 in do_global_ctors at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:104
      (inlined by) start_cpu0_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:206
  #5  0x403769b6 in call_start_cpu0 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/cpu_start.c:849 (discriminator 1)

````

This seems to be different from  https://github.com/m5stack/M5Unified/issues/89  and https://github.com/m5stack/M5Unified/issues/121  - it points to an issue with the I2C driver version

unfortunately to use M5Unified [I2S sound with Arduino 3.x, rc2 is required ](https://github.com/espressif/arduino-esp32/issues/11004)

thanks for having a look 

Michael

lovyan03, 2025-03-25:
Hello, @mhaberler 
I just update M5GFX develop branch, please try it.

mhaberler, 2025-03-25:
excellent - works fine for me!

Here's a QRcode reading demo using the M5Stack CoreS3 builtin camera - works great with your patch and Arduino 3.2rc2!

https://github.com/mhaberler/cores3-read-qrcode

thanks a lot

Michael

closed 2025-03-25 by mhaberler