m5stack/M5Unified #159: Helloworld example broken?

jesusfj710, 2025-03-26:
In [here](https://docs.m5stack.com/en/arduino/m5unified/helloworld), if you try to compile it, you get an error. I'm pretty sure that this is due that the `Display` in the loop is not using `M5` namespace, right?

[image]

```
/tmp/.arduinoIDE-unsaved2025226-11115-8mja3h.rnl4w/sketch_mar26b/sketch_mar26b.ino: In function 'void loop()':
/tmp/.arduinoIDE-unsaved2025226-11115-8mja3h.rnl4w/sketch_mar26b/sketch_mar26b.ino:28:3: error: 'Display' was not declared in this scope
   Display.setCursor(0, 20);              // set character drawing coordinates (cursor position)
   ^~~~~~~

exit status 1

Compilation error: 'Display' was not declared in this scope
```

By fixing the loop with the following code, i'm able to get it to compile. (as well adding the right `Serial.begin(...)` in the setup.
```
void loop() {

  M5.Display.setCursor(0, 20);              // set character drawing coordinates (cursor position)
  M5.Display.printf("COUNT: %d\n", count);  // display count on screen
  Serial.printf("COUNT: %d\n", count);   // display count serially
  count++;                               // increase count by 1
  delay(1000);                           // wait 1 second(1,000msec)
}
```

[image]

However, nothing is displayed in my PaperS3. Why? This are my settings

[image]

[image]

[image]

I did also clone the epdiy repo to the libraries folder in Arduino.

If I tried with the ESP32S3 DevModule from Espressif with an the AnalogBar examples, including the epdiy lib, works...

There is no way I can use the M5Unified lib with ESP32S3 dev module, and there is no way I can use the display without the ESP32S3 dev module.

jesusfj710, 2025-04-10:
> have you tried adding #include "epdiy.h"

Yes, I did. And another problem appears when you do that. It doesn't seem that the latest version of epdiy.h works correctly with M5Unified.

lovyan03, 2025-04-17:
Hello @jesusfj710 , @RikerZhu 
I apologize for the long wait.

I have updated the M5GFX `develop` branch.
The new version works with M5GFX alone, without EPDiy.
Please try it out.
If there are no problems, it will be included in the next update release.

jesusfj710, 2025-05-01:
I will close this issue as this is more related to the doc in the helloworld example and its already rectified.

closed 2025-05-01 by jesusfj710