m5stack/M5Unified #163: Possible to disable I2C or update to new master/slave driver?

Sean-Der, 2025-05-17:
I am trying to use esp_camera and M5Unified but that results in this error.

`E (2027) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver`

What is the best way to use M5Unified in esp-idf and capture the camera?

mhaberler, 2025-05-17:
try the develop branch
```ìni
[env]
lib_deps =
        https://github.com/m5stack/M5Unified.git#develop
```

Sean-Der, 2025-05-18:
Thanks for the help @mhaberler that didn't fix it for me. I am using esp-idf, so maybe a different behavior vs platform.io?

I was able to fix it by pinning esp-camera to 2.0.10. In the subsequent release they changed to new i2c API. So things work for me! 

I will leave this issue open, but feel please close if the upgrade isn't planned or wanted :) So much is happening with Espressif + software I never sure if I am doing the right thing

ainyan03, 2026-08-27:
Following up on this one, because it changed with the recent releases.

With M5GFX 0.2.28 / M5Unified 0.2.21 on ESP-IDF 5.3.2 or later, `M5.In_I2C` / `M5.Ex_I2C` open their port through the ESP-IDF `i2c_master` driver (`i2c_new_master_bus`), and a port that the application (or another component) has already opened with that driver is detected and shared instead of being re-initialised. So the old `CONFLICT! driver_ng is not allowed to be used with this old driver` situation no longer applies, and there is no need to pin esp32-camera to 2.0.10.

To share the sensor's I2C bus with esp32-camera 2.0.11 or later, let it reuse the port M5Unified already opened instead of opening its own:

```cpp
camera_config_t config = {};
config.pin_sccb_sda = -1;                 // do not open a new bus
config.pin_sccb_scl = -1;
config.sccb_i2c_port = M5.In_I2C.getPort();   // the port M5Unified uses for the PMIC/IMU/camera
```

esp32-camera then fetches the existing bus handle with `i2c_master_get_bus_handle()` (its `SCCB_Use_Port()` path). The reverse order — `esp_camera_init()` first, `M5.begin()` afterwards — is also handled since M5GFX 0.2.28.

This is from the code and from the I2C-sharing tests done for that release, not from running esp32-camera itself here; if it still conflicts for you with the current versions, please post the log.

ainyan03, 2026-08-28:
Sorry for the long silence on this one. Short version: with the current versions esp32-camera and M5Unified can share the I2C bus, and there is no need to pin esp32-camera to 2.0.10 any more. One remaining problem on the released M5Unified 0.2.21 has just been fixed in `develop` (#345) and will be in the next release; until then please use the `develop` branch, or call `M5.begin()` before `esp_camera_init()` (details below).

I have now actually run esp32-camera together with M5Unified on a CoreS3 (GC0308) to confirm this: ESP-IDF 5.5.1, esp32-camera 2.1.7, M5GFX 0.2.28, M5Unified `develop`.

### Recommended order: `M5.begin()` first, then the camera on the same port

```cpp
auto cfg = M5.config();
M5.begin(cfg);

camera_config_t config = {};
// ... pins / format as usual ...
config.pin_sccb_sda = -1;                     // do not open a new bus
config.pin_sccb_scl = -1;
config.sccb_i2c_port = M5.In_I2C.getPort();   // the port M5Unified uses for the PMIC / IMU / camera
esp_camera_init(&config);
```

esp32-camera (2.0.11 or later) then fetches the existing bus handle with `i2c_master_get_bus_handle()` (its `SCCB_Use_Port()` path). Tested: the sensor is detected, QVGA frames keep streaming, and reading the PMIC through `M5.In_I2C` on the same bus keeps working at the same time. This order works on the released 0.2.21 as well.

### Why the conflict is gone

Since M5GFX 0.2.28 / M5Unified 0.2.21 (ESP-IDF 5.3.2 or later), `M5.In_I2C` / `M5.Ex_I2C` open their port through the ESP-IDF `i2c_master` driver (`i2c_new_master_bus`), and a port that the application or another component has already opened with that driver is detected and shared instead of being re-initialised. The old `CONFLICT! driver_ng is not allowed to be used with this old driver` situation no longer applies.

### The reverse order (`esp_camera_init()` first) needs `develop` / the next release

The I2C side is fine in this order too (M5GFX picks up the bus esp32-camera opened), but on M5Unified 0.2.21 `M5.begin()` drives GPIO46 as an output during board detection (it is the power-hold pin of some other ESP32-S3 boards), and on CoreS3 that pin is the camera VSYNC, so the frames stop. #345 restores the pin once the board is known; with it applied both orders stream frames.

ainyan03, 2026-09-14:
A follow-up: the remaining problem (#345, GPIO46 restored after board detection) has just been released as **M5Unified 0.2.22** (with **M5GFX 0.2.29**). With the released versions, esp32-camera and M5Unified share the I2C bus as described above, and there is no need to use the `develop` branch any more.

I believe this issue can be closed.