m5stack/M5Unified #165: M5Unified 0.2.6 causes compilation issues with M5Dial

avancells, 2025-05-17:
Hi,

After a really frustrating morning trying to do the initial setup for the M5Dial, I have realized that using the latest version (Released may 16th 2025) causes compilation issues when using the M5Dial package.

Unless the version is pinned to m5stack/M5Unified@0.2.5 it will fail to compile with errors like:
```
In file included from .pio/libdeps/m5stack-stamps3/M5Unified/src/utility/Mic_Class.cpp:29:
C:/Users/XXXXX/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/hal/esp32s3/include/hal/i2s_ll.h: In function 'void i2s_ll_tx_set_active_chan_mask(i2s_dev_t*, uint32_t)':
C:/Users/XXXXX/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/hal/esp32s3/include/hal/i2s_ll.h:484:48: error: no matching function for call to '$_64(volatile i2s_dev_s::<unnamed union>&)'
     typeof(hw->tx_tdm_ctrl) tdm_ctrl_reg = hw->tx_tdm_ctrl;
                                                ^~~~~~~~~~~
In file included from .pio/libdeps/m5stack-stamps3/M5Unified/src/utility/Mic_Class.hpp:14,
                 from .pio/libdeps/m5stack-stamps3/M5Unified/src/utility/Mic_Class.cpp:4:

C:/Users/XXXXX/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/i2s_struct.h:244:11: note: candidate: 'constexpr i2s_dev_s::<unnamed union>::<constructor>(const i2s_dev_s::<unnamed 
union>&)' <near match>
     union {
           ^
C:/Users/XXXXX/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/i2s_struct.h:244:11: note:   conversion of argument 1 would be ill-formed:
In file included from .pio/libdeps/m5stack-stamps3/M5Unified/src/utility/Mic_Class.cpp:29:
C:/Users/XXXXX/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/hal/esp32s3/include/hal/i2s_ll.h:484:48: error: binding reference of type 'const i2s_dev_s::<unnamed union>&' to 'volatile i2s_dev_s::<unnamed union>' discards qualifiers
     typeof(hw->tx_tdm_ctrl) tdm_ctrl_reg = hw->tx_tdm_ctrl;
                                            ~~~~^~~~~~~~~~~
In file included from .pio/libdeps/m5stack-stamps3/M5Unified/src/utility/Mic_Class.hpp:14,
                 from .pio/libdeps/m5stack-stamps3/M5Unified/src/utility/Mic_Class.cpp:4:

```

Note that I am using platformio, but this issue also happens compiling through Arduino IDE.

I think there must be something wrong either from the M5Dial library or M5Unified. If M5Dial is not compatible with versions >0.2.5 then it should be stated in the M5Dial documentation, tutorials, etc.

lovyan03, 2025-05-17:
Hello, @avancells 
sorry. I just updated v0.2.7. please try this.

avancells, 2025-05-17:
Just tested with 0.2.7 and it is compiling properly!

For awareness, this issue might still be faced by any user depending on the library managers (Arduino IDE's, Platformio, etc.) if they have their dependency pinned to the @latest version