m5stack/M5Unified #176: touch lib problem with arduino core for esp32 v3.3.0

GrandeRipper, 2025-08-04:
I use the current Arduino core for the ESP32 v3.3.0. and M5Unified also in the current version 0.2.7
with the Arduino core for the ESP32 v3.2.1 everything still worked fine. Since the update to the new version, Arduino now produces the following error:

\Arduino\libraries\M5Unified\src\M5Unified.cpp: In member function 'm5::board_t m5::M5Unified::_check_boardtype(m5::board_t)':
c:\Users\MAX\Documents\Arduino\libraries\M5Unified\src\M5Unified.cpp:755:15: error: 'touch_pad_init' was not declared in this scope; did you mean 'touch_pad_t'?
  755 | touch_pad_init();
 | ^~~~~~~~~~~~~~
 | touch_pad_t
c:\Users\MAX\Documents\Arduino\libraries\M5Unified\src\M5Unified.cpp:756:15: error: 'touch_pad_config' was not declared in this scope; did you mean 'touch_pad_t'?
  756 | touch_pad_config(TOUCH_PAD_NUM4, TOUCH_PAD_THRESHOLD_MAX); // TOUCH_PAD_NUM4 == GPIO13
 | ^~~~~~~~~~~~~~~~
 | touch_pad_t

and so on...

**This can be tested by trying to compile the standard example SliderUI !**

If you add the following lines to M5Unified.hpp the code works but warnings are displayed during compilation.

```
#if __has_include(‘driver/touch_pad.h’)
  #include ‘driver/touch_pad.h’
#endif
```

Warning:

In file included from C:\Users\MAX\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.5-b66b5448-v1\esp32/include/driver/touch_sensor/include/driver/touch_pad.h:12,
                 from c:/Users/MAX/Documents/Arduino/libraries/M5Unified/src/M5Unified.hpp:14,
                 from c:\Users\MAX\Documents\Arduino\libraries\M5Unified\src\utility\Power_Class.cpp:6:
C:\Users\MAX\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.5-b66b5448-v1\esp32/include/driver/touch_sensor/esp32/include/driver/touch_sensor.h:13:2: warning: #warning "This set of Touch APIs has been deprecated, please include 'driver/touch_sens.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress legacy driver deprecated warning' option under 'Touch Configuration' menu in Kconfig" [-Wcpp]
   13 | #warning "This set of Touch APIs has been deprecated, \
      |  ^~~~~~~
In file included from C:\Users\MAX\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.5-b66b5448-v1\esp32/include/driver/touch_sensor/include/driver/touch_pad.h:12,
                 from c:/Users/MAX/Documents/Arduino/libraries/M5Unified/src/M5Unified.hpp:14,
                 from c:\Users\MAX\Documents\Arduino\libraries\M5Unified\src\utility\Speaker_Class.cpp:6:
C:\Users\MAX\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.5-b66b5448-v1\esp32/include/driver/touch_sensor/esp32/include/driver/touch_sensor.h:13:2: warning: #warning "This set of Touch APIs has been deprecated, please include 'driver/touch_sens.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress legacy driver deprecated warning' option under 'Touch Configuration' menu in Kconfig" [-Wcpp]
   13 | #warning "This set of Touch APIs has been deprecated, \

ainyan03, 2026-08-17:
Thank you for your patience — this compile error was fixed in v0.2.8, and the current release 0.2.20 keeps the new/legacy touch-driver switching with further hardening on top. We also confirmed a clean build against Arduino core 3.3.4. Please confirm your build passes with 0.2.20; I believe this can then be closed.

GrandeRipper, 2026-08-19:
I can confirm that everything is now working perfectly.
Thank you very much!

closed 2026-08-19 by GrandeRipper