m5stack/M5Unified #180: Tab5 : M5.Power.powerOff() creates screen burns

tobozo, 2025-08-26:
Hi,

This Tab5 was turned off last night using `M5.Power.powerOff()` instead of using the power button. 
When turning it on this morning, the display shows traces of screen burn on the borders.

[image]

`M5.Power.powerOff()` is expected to power everything off, right?
The ESP32P4 and CDC are effectively turned off, but it seems like the display stays on.

tobozo, 2025-08-26:
obviously `tft.fillScreen(TFT_WHITE);delay(1200);` fixes the screen burn, but the issue is more about `Power.powerOff()` behaviour

lovyan03, 2025-08-26:
Hello, @tobozo 
Thank you for the information. and I apologize for any inconvenience caused.
I have now updated the develop branch and adjusted it so that the power can be shut down.