m5stack/M5Unified #181: Sample applications mostly does not work with CoreS3

fariouche, 2025-08-27:
Hello!

I'm trying to check an example with audio output on my CoreS3... So I tried arduino IDE with esp32 library 3.3.0 and M5Unified 0.2.7.

None of the sample apps are working!!
The only sample I was able to make work is "game of life"...
In my case, I pin pointed that M5.begin() make the device crash. I tried to add the M5.begin() at the start of game of life and it was not working anymore too.

This is very frustrating. I'm totally unable to use my CoreS3!
By the way, same issue with UIFlow too. Now I understand that this is because of M5Unified that is broken.

To reproduce just compile and upload any sample, for example the Speaker sample app.
I'm then unable to upload any example without unplugging the device, plug it back and quickly press the reset button (if I wait too long after plugging in, I cannot enter programming mode even if I press the reset button for a very long time)

I tried to use the debugger, but it fails too... except if used with the game of life example.

I also tried with platformio with https://github.com/m5stack/CoreS3-UserDemo, but it fails too (black screen, no more ttyACM0 etc)
Oh, I forgot: I'm using linux (not sure this is important)

Do you have any idea of the root cause?

Let me know if you need me to do some more tests

Thanks!

fariouche, 2025-08-27:
Some news:
I tried M5Burner...
CoreS3 User Demo  (v0.6) fails (same with v0.4)
CoreS3 touch panel hot-fix v0.5 fails
CoreS3 box demo v0.1: works! (and I have sound) (so my unit is not faulty)

lovyan03, 2025-08-28:
Hello, @fariouche 

Are you using ArduinoESP32 v3.3.0?
If so, try downgrading to v3.2.1 or lower, or using the develop branch of M5GFX and M5Unified.

And, I'm sorry, I don't understand English, so I can't accurately understand what you mean when you explain it in writing.

Also, which game of life sample are you referring to? Is the sample you are trying the M5GFX one?

Please specify exactly which samples worked and which didn't, including the URLs.

fariouche, 2025-08-28:
Hello @lovyan03 for your answer!
I finally understood the problem: M5Unified is powering off the device and I must press the power button at least once!
Now everything is working.

Thank you!

closed 2025-08-28 by fariouche