m5stack/M5Unified #185: TAB5 RTC AND TOUCH

fermintm, 2025-09-14:
I'm having trouble using the RTC and Touch in the same code. When using the RTC, the Touch stops working, and other times the RTC fails altogether. I think it's due to the wire.begin function, which breaks communication between devices. I'm using the latest M5GFX and M5Unified libraries. There's no example for the internal clock on the board's docs page.

The board has been on the market for months, and fully functional Arduino libraries have yet to be created. Please, before listing modules or cores, also include everything necessary for buyers.

tobozo, 2025-09-17:
Hi, 

I didn't know there was support for RTC8130 in M5Unified, how did you enable it?

I had to create a custom version in a tab5 project with support for read/writes to RTC RAM. 
It's not 100% compatible with M5Unified's RTC API (still needs a couple of methods) but it reuses `m5::I2C_Class` and doesn't exhibit any conflict as long as RTC and Touch are polled from the same core.

https://github.com/tobozo/M5Tab5-Game-and-Watch/blob/main/gw/m5unified-rtc.hpp 

```cpp
m5::RTC8130_Class Tab5Rtc;
// Tab5Rtc.begin();
```

fermintm, 2025-09-17:
Sorry. I couldn't explain myself. For RTC, I used another library extracted from code for Tab5. The problem is that when using Wire.begin(), the touch stops working. Supposedly, if the internal I2C devices use the same data line, this problem shouldn't arise. I'm still waiting for M5stack to compile all the libraries for this board on Arduino. I don't understand much, and I'm not going to learn ESP-IDF anymore.

tobozo, 2025-09-17:
When using `m5::In_I2C` (the `m5::I2C_Class` singleton shared with touch), you don't need to use `Wire.begin()` because `M5.begin()` already took care of that when init'ing the display and touch.

Moreover, no esp-idf or Arduino Wire functions are required, M5Unified I2C drivers use one of the two `m5::I2C_Class` abstractions, see the [RTC8563_Class](https://github.com/m5stack/M5Unified/blob/224c1dfac14a05afd201972542c5768094704184/src/utility/RTC8563_Class.cpp#L43-L54) where the equivalent of `Wire.begin()` happens.

Note: RTC8563 is *not* for Tab5, it only holds some [interesting types and functions](https://github.com/m5stack/M5Unified/blob/224c1dfac14a05afd201972542c5768094704184/src/utility/RTC8563_Class.cpp#L10-L41) needed by M5Unified which could benefit additional RTC drivers.

fermintm, 2025-09-17:
> When using `m5::In_I2C` (the `m5::I2C_Class` singleton shared with touch), you don't need to use `Wire.begin()` because `M5.begin()` already took care of that when init'ing the display and touch. [...]

https://github.com/fermintm/Tab5-DEV-CODE

tobozo, 2025-09-27:
[v0.2.9](https://github.com/m5stack/M5Unified/releases/tag/0.2.9) now implements a RTC driver for Tab5 `RX8563_Class`, the interesting types and functions have been moved into a RTC base class and are now inherited by two RTC drivers.

The Tab5 RTC implementation doesn't have methods to read/write the RTC RAM though, let's hop this eventually gets added.

fermintm, 2025-09-27:
> [v0.2.9](https://github.com/m5stack/M5Unified/releases/tag/0.2.9) now implements a RTC driver for Tab5 `RX8563_Class`, the interesting types and functions have been moved into a RTC base class and are now inherited by two RTC drivers.
> 
> The Tab5 RTC implementation doesn't have methods to read/write the RTC RAM though, let's hope this eventually gets added.

system crash.

RTC found.
Guru Meditation Error: Core  1 panic'ed (Load access fault). Exception was unhandled.

Core  1 register dump:
MEPC    : 0x4fc1b1e4  RA      : 0x400511d8  SP      : 0x4ff1d910  GP      : 0x4ff11300  
TP      : 0x4ff1df60  T0      : 0x4fc10cc4  T1      : 0x20000000  T2      : 0x00000009  
S0/FP   : 0x00000000  S1      : 0x00000000  A0      : 0x00000000  A1      : 0xffffffff  
A2      : 0x00000004  A3      : 0x7f7f7f7f  A4      : 0x00000000  A5      : 0x00000004  
A6      : 0xa0000000  A7      : 0x0000000a  S2      : 0x40078000  S3      : 0x00000003  
S4      : 0xffffffff  S5      : 0x00000004  S6      : 0x4ff1da34  S7      : 0x00000073  
S8      : 0x00000003  S9      : 0x00000000  S10     : 0x00000003  S11     : 0x00000002  
T3      : 0x00000000  T4      : 0xffff01ff  T5      : 0x19999999  T6      : 0x00000019  
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000005  MTVAL   : 0x00000000  
MHARTID : 0x00000001

fermintm, 2025-09-28:
Arduino IDE 2.3.6. Latest M5Unified and M5GFX. Latest M5stack board manager. and RTC example include in m5unified.